arduino/arduino-pro-ide #96: i can see my installed board but i can't see any port

Master811129, 2019-10-24:
**Describe the bug**
i can see my installed boards but i can't see any COM port in IDE.
v0.0.1-alpha.preview-snapshot-20191021.1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Arduino Pro IDE'
2. Click on 'boards and ports'
4. See error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10 enterprise]

mastrolinux, 2019-10-25:
Should be fixed by https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview

Master811129, 2019-10-26:
> Should be fixed by https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview

yeah it fixed thank you🎉🎉🌭

closed 2019-10-26 by Master811129