arduino/arduino-pro-ide #101: Sketch is lost when changing from classic to adva

sstaub, 2019-10-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Master811129, 2019-10-26:
yeah i also see it in windows 10.

kittaakos, 2019-11-12:
Thanks for reporting it, @sstaub.

This is a glitch we are working on; the editors are not restored between the sessions. As a workaround, you can click on the `Explorer` icon on the left side-bar (or <kbd>Ctrl/Cmd</kbd>+<kbd>Shift</kbd>+<kbd>E</kbd>) and open your sketch file manually. Please reference the attached screencast:

[image]

kittaakos, 2019-11-12:
Upstream issue: https://github.com/eclipse-theia/theia/issues/6530

closed 2019-12-09 by kittaakos