arduino/arduino-pro-ide #99: Could not find Git on the PATH.

iotsharer, 2019-10-25:
I can use git on shell,but when I open the arduino-pro-ide.It say Could not find Git on the PATH.
Why? And how to do ?(My OS is Win10 1903)
[image]

[image]

kittaakos, 2019-10-28:
@iotsharer, where is your Git installed on Windows? This [issue](https://github.com/TypeFox/find-git-exec/issues/11) could be related. Thanks!

iotsharer, 2019-10-28:
> @iotsharer, where is your Git installed on Windows? This [issue](https://github.com/TypeFox/find-git-exec/issues/11) could be related. Thanks!

I installed it in D:/Program Files/Git/ and I have set it to system Path.I can use git on powershell
[image]
[image]

kittaakos, 2019-10-28:
Thanks for the details.

tigoe, 2019-11-06:
Got this same issue on MacOS 10.14.6. The message showed up in a toast at the bottom of the screen and I can't make it go away. There is no indication in the UI of how to solve the issue.

parasquid, 2019-11-17:
Just installed on Windows 10 and had the same toast message (found this issue after following the google links). I am however able to "make it go away" by clicking on the `x` mark in the toast.

Not sure what it's affecting though since I seem to be able to add board information just fine.

My git is found via:
```
git --info-path
C:/Users/tristan/scoop/apps/git/2.24.0.windows.2/mingw64/share/info
```

kittaakos, 2019-11-22:
This has been resolved in Theia, we wait for npmjs to fix [an HTTP 503 issue](https://status.npmjs.org/incidents/gtjlhzsjqsyr) on their side.

kittaakos, 2019-11-25:
This will be available in the next release.

jankeromnes, 2019-12-12:
- [x] https://github.com/TypeFox/find-git-exec/issues/11 has been fixed
- [x] `find-git-exec` has been updated in Theia: https://github.com/eclipse-theia/theia/pull/6602
- [x] Theia has been updated in Arduino Pro IDE

I.e. this should be fixed now. Thanks for the great work!

closed 2019-12-12 by jankeromnes