arduino/arduino-pro-ide #103: "Additional Boards Manager URL" cannot be found in Preferences?

uncle-yong, 2019-10-26:
Version: arduino-pro-ide-0.0.2-alpha.preview-windows

I could not test 3rd party ESP8266 boards too inside. It could be good if we have that option to install 3rd party boards like in this JSON file: http://arduino.esp8266.com/stable/package_esp8266com_index.json

thetinker11, 2020-01-11:
there's a workaround on adding custom boards by installing it in Arduino IDE, with those custom board links. After that, install arduino pro ide.

-works on linux as of now

toogooda, 2020-02-10:
> there's a workaround on adding custom boards by installing it in Arduino IDE, with those custom board links. After that, install arduino pro ide.
> 
> -works on linux as of now

Also Windows10 but doesn't work with search

infinitymakerspace, 2020-10-05:
This workaround does not work

per1234, 2020-10-05:
I'm happy to tell you that there is no longer any need for a workaround. Arduino Pro IDE now has support for installing 3rd party boards platforms via Boards Manager. Please see the instructions here:
https://github.com/arduino/arduino-pro-ide/issues/245

kittaakos, 2021-02-11:
The most recent way of installing 3rd party boards is the same as with the Java IDE. Closing.

closed 2021-02-11 by kittaakos