arduino/arduino-pro-ide #104: Missing Processor selection option

rnbokade, 2019-10-26:
The processor selection option e.g. the one that used to come when Arduino pro mini used to be selected is missing in this. Please add it to the release soon. I don't think it would be that hard

Erriez, 2019-10-28:
Yes, the `board` and `processor` menu's are missing in in new IDE v0.0.2 Alpha:
[image]

Original Arduino IDE:
[image]

Now I can't use my favorite `Pro or Pro Mini 3V3 8MHz board` and my custom boards with the new IDE.

HiFiJ, 2020-04-01:
Hi everyone,

I am also missing this option; out of luck for now with my Pro Mini 3.3V boards. 
Thanks for the work on the new IDE in general, would love to use it!

Master811129, 2020-04-01:
> Yes, the `board` and `processor` menu's are missing in in new IDE v0.0.2 Alpha:
> [image]
> 
> Original Arduino IDE:
> [image]
> 
> Now I can't use my favorite `Pro or Pro Mini 3V3 8MHz board` and my custom boards with the new IDE.

I think this is #10. It'll be available in the next release

per1234, 2020-04-02:
Closing as duplicate of #10

Thanks @Master811129!

closed 2020-04-02 by per1234