arduino/arduino-pro-ide #105: Opening Bracket Bug

TheMindVirus, 2019-10-26:
Occasionally when I type a bracket, a text selection from the beginning of the code to the current position happens automatically. When typing the next character, it deletes that entire section of code. This is extremely frustrating. The workaround at the moment is having to press Ctrl-Z to undo each time it happens. I tried disabling autosuggest and autoclosebrackets but it still occurred.

Love the new look IDE!

TheMindVirus, 2019-10-26:
Shift is sometimes a little slow to react. These are issues which may be fixed by the performance gains of moving to C++.

spoenemann, 2020-01-13:
What keyboard layout / language do you use?

TheMindVirus, 2020-05-14:
Either English US or English UK, most likely UK.

spoenemann, 2020-05-15:
Maybe related: https://github.com/eclipse-theia/theia/pull/6881 and https://github.com/eclipse-theia/theia/issues/7315

@TheMindVirus could you check whether the problem still occurs with the latest version of Pro IDE?

TheMindVirus, 2020-06-13:
The latest version seems to be very slick and Visual Studio based with IntelliSense autocomplete.
I haven't seen the bug crop up in a while, I'm going to assume it has been fixed.

spoenemann, 2020-06-15:
Ok, please reopen if it occurs again.

closed 2020-06-15 by spoenemann