arduino/arduino-pro-ide #106: "pins_arduino" not found

maxerd, 2019-10-26:
everytime i try to verify a programme I get the error saying "Error: Request compile failed with message: build failed: exit status 1"
when i look at "Arduino: daemon error" this comes up;

[INFO] In file included from C:\Users\maxja\AppData\Local\Temp\arduino-sketch-582A576645E1F7EDE1CA6E4439C98AD0\sketch\sketch_october26a.ino.cpp:1:0:
[INFO] C:\Users\maxja\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.1\cores\arduino/Arduino.h:257:10: fatal error: pins_arduino.h: No such file or directory
[INFO]  #include "pins_arduino.h"
[INFO]           ^~~~~~~~~~~~~~~~
[INFO] compilation terminated.

because of this i cant code or upload anything .

I expect its something that is wrong on my pc or on my own part because i have not seen anyone on here that has reported the some problem. 

[image]
the image above is what i get when i open the programme(i got the error for the first time with v0.0.2)

[image]
the image above is error as shown in "Arduino: daemon error"

on windows 10

mastrolinux, 2019-10-29:
Your file seems huge, it looks like you are including the Arduino.h file in your own sketch overriding our own official one. Do you mind removing it and trying again?

maxerd, 2019-10-29:
that was something i was testing,  should have removed that before uploading it here. But had that problem before it, here are the screenshots without that part of the programme.

[image]

[image]

mastrolinux, 2019-11-26:
This should be solved with the last version of the CLI, please try again after installing it.

closed 2021-02-18 by per1234