arduino/arduino-pro-ide #107: Cannot access resource at

marcos-guillen, 2019-10-26:
Cannot access resource at /d:/Desa/Arduino/arduino-pro-ide-0.0.2-alpha.preview-windows/work1.theia-workspace.

Arduino:
[Error - 15:57:01] Request textDocument/codeAction failed.
Error: Document is not available: file:///d%3A/Desa/Arduino/ejercicios/ARBOL/Alvaro/practica1/practica1.ino
    at handleResponse (file:///D:/Desa/Arduino/arduino-pro-ide-0.0.2-alpha.preview-windows/resources/app/lib/bundle.js:123758:36)
    at processMessageQueue (file:///D:/Desa/Arduino/arduino-pro-ide-0.0.2-alpha.preview-windows/resources/app/lib/bundle.js:123582:9)
    at Immediate.<anonymous> (file:///D:/Desa/Arduino/arduino-pro-ide-0.0.2-alpha.preview-windows/resources/app/lib/bundle.js:123565:7)
    at runCallback (timers.js:694:18)
    at tryOnImmediate (timers.js:665:5)
    at processImmediate (timers.js:647:5)

spoenemann, 2019-12-16:
Any hints to reproduce this?

per1234, 2021-02-18:
Hi @marcos-guillen. Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234