arduino/arduino-pro-ide #109: Not uploading to Uno - intermittent.

Durakill, 2019-10-27:
Sketch compiles fine, without error.

Uploading to Uno appears to take place without error.

However, sketch has not uploaded.

Unfortunately, problem is intermittent. 

Just now, I disconnected the Uno, and reconnected it. The board is now communicating - sketch downloaded, and serial monitor is reporting data. However, I think this fix is also intermittent - sometimes works, sometimes not. 

I am happy enough to send you log files or similar if you wish.

Thanks,

Michael Flynn

OS X 10.14.6

kittaakos, 2019-11-26:
> Sketch compiles fine,

@Durakill, can you please share your sketch file?

kittaakos, 2019-12-09:
> I am happy enough to send you log files or similar if you wish.

Please do so.

Without the logs, I've assumed the following use-case: due to a bug in the CLI we were not able to close the serial monitor port and disconnect, hence the upload failed because the serial monitor was using the port. This issue has been fixed in the  CLI (see #157) and I have completely rewritten the serial-monitor feature. (see #112)

Durakill, 2019-12-09:
Hi,

The problem has not recurred after downloading a more recent version the IDE. Thank you for your efforts. 

Regards, 

Michael Flynn

kittaakos, 2019-12-10:
Thank you for the update, I am going to close this issue.

Please reopen if required.

closed 2019-12-10 by kittaakos