arduino/arduino-pro-ide #111: How to Setup on Debian Stretch?

harryberlin, 2019-10-27:
Tried the Linux Version. After running the arduino-electr.... file, there was only the sandclock and than nothing.

mastrolinux, 2019-10-28:
Please try the latest v0.0.2 version and let me know

harryberlin, 2019-10-28:
where to find the how to?

cmaglie, 2019-10-30:
https://github.com/arduino/arduino-pro-ide/releases

harryberlin, 2019-11-02:
there is only the release download. no manual for install on debian.

tried the windows version:
folder loading is ok.
after opening i get some error notifications. something like failed to start language server, unexpected git executable.
couldn't start the client arduino.
the ide forget the selected board.
i have installed the library SimpleTimer but the compiler shows lot of errors.
how to install libraries from zip?
how to select the nano bootloader versions?
at the end:
i like the old ide :)

closed 2020-09-24 by per1234