arduino/arduino-pro-ide #112: The serial port does not work

baghjanati, 2019-10-27:
The serial port does not send or receive information

mastrolinux, 2019-10-28:
I am not able to reproduce. Please send us a video or give more info.

stevenloppe, 2019-11-25:
I am encountering a similar problem.
The code I am executing is:
```
void setup() {
  Serial.begin(9600);
  digitalWrite(LED_BUILTIN, HIGH); 
}
void loop() {
    Serial.println("This is a test");
    Serial.write(98);
    delay(2000);
}
```

With the original IDE if I open this file/project, select Uno at COM6, upload, and open the serial monitor I will see "This is a test" and "b" printed.

However if I do the same steps with this ide, open the workspace/folder, select Uno and COM6, upload, and then open the serial monitor I will NOT see any output.

I am running v0.0.2 on Windows 10.

If it is worth anything I am seeing the same error as #107 in the "Arduino" output window, it seems to go away when this ide is run as administrator.

kittaakos, 2019-11-26:
Thank you for the snippet, @stevenloppe 

> same error as #107

It comes from the Arduino language server and completely unrelated.

kittaakos, 2019-11-28:
@stevenloppe, I started the PRO IDE form the HEAD of the source, and I can confirm, it works as expected:
[image]

I am going to try it with the `0.0.2` release on Windows right now.

kittaakos, 2019-11-28:
Great, I could reproduce it. it is definitely a UI issue as I see in the messages in the logs.

kittaakos, 2019-11-28:
> I could reproduce it.

Steps to reproduce:
 - Start of the application,
 - verify and upload the [sketch](https://github.com/arduino/arduino-pro-ide/issues/112#issuecomment-557946013),
 - open the serial monitor you can see the messages,
 - stop the app,
 - start the app, you cannot see the messages but they're received according to the logs.

kittaakos, 2019-11-28:
I have found another bug which is related to the serial-monitor; if I start the app and attach a board that runs the serial-monitor sketch, then I cannot upload:
```
avrdude: ser_open(): can't open device "/dev/cu.usbmodem14401": Resource busy
Upload error: Error: uploading error: exit status 1
```

kittaakos, 2019-11-28:
As I see, there are a few issues here; start the app, try to send anything to the connected board:
```
root ERROR Request send failed with error: No connection with ID: ece0f433-8844-48f1-894b-4dcbf2b71bf7. Error: No connection with ID: ece0f433-8844-48f1-894b-4dcbf2b71bf7.
    at MonitorServiceImpl.<anonymous> (/Users/akos.kitta/git/arduino-editor/arduino-ide-extension/lib/node/monitor/monitor-service-impl.js:238:27)
    at step (/Users/akos.kitta/git/arduino-editor/arduino-ide-extension/lib/node/monitor/monitor-service-impl.js:41:23)
    at Object.next (/Users/akos.kitta/git/arduino-editor/arduino-ide-extension/lib/node/monitor/monitor-service-impl.js:22:53)
    at /Users/akos.kitta/git/arduino-editor/arduino-ide-extension/lib/node/monitor/monitor-service-impl.js:16:71
    at new Promise (<anonymous>)
    at __awaiter (/Users/akos.kitta/git/arduino-editor/arduino-ide-extension/lib/node/monitor/monitor-service-impl.js:12:12)
    at MonitorServiceImpl.send (/Users/akos.kitta/git/arduino-editor/arduino-ide-extension/lib/node/monitor/monitor-service-impl.js:228:16)
    at JsonRpcProxyFactory.<anonymous> (/Users/akos.kitta/git/arduino-editor/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:218:73)
    at step (/Users/akos.kitta/git/arduino-editor/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:47:23)
    at Object.next (/Users/akos.kitta/git/arduino-editor/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:28:53)
monitor-service WARN Could not find monitor client for connection ID: ece0f433-8844-48f1-894b-4dcbf2b71bf7
```

kittaakos, 2019-11-28:
Timestamps are also broken:
```
This is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
14:02:27.380 -> bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
bThis is a test
14:02:33.392 -> bThis is a test
bThis is a test
bThis is a test
14:02:39.406 -> bThis is a test
bThis is a test
bThis is a test
```
I received only two timestamps, although it was enabled all the time.

kittaakos, 2019-12-09:
I have completely rewritten the serial-monitor feature; the fix will be available with the next release.

closed 2019-12-11 by kittaakos