arduino/arduino-pro-ide #114: Confirm Exit not working

djoel1, 2019-10-27:
Preference Application ConfirmExit   tried always and ifRequired
Push exit, application closes without asking anything

**Expected behavior**
Likley show a message box asking you if you want to exit

**Desktop (please complete the following information):**
 - OS: Windows 7
  IDE 0.0.2

mastrolinux, 2019-10-28:
This one was a bug in Theia https://github.com/eclipse-theia/theia/blob/aff11bec2a0424a63e506613d9e23f2e6bdd1d42/packages/core/src/browser/window/default-window-service.ts 

Will be fixed when we will rebase to the new master.

kittaakos, 2019-11-05:
I have updated to `next` (`0.13.0-next.f2b35569`) Theia; I can confirm it works now on macOS. I will re-verify this item before roll out our next release.

kittaakos, 2019-12-11:
Was fixed in `0.0.3`.

closed 2019-12-11 by kittaakos