arduino/arduino-pro-ide #115: questions about serial plotter.

breaker123, 2019-10-28:
**Is your feature request related to a problem? Please describe.**
Hi, Awesome work! well done.I have used Arduino for 2 years,I have made many interesting projects.I wanna thank your team with fantastic work.
But ,I also met questions, When I use the serial plotter, I want to extend the display range of the horizontal axis time so that the latest data acquisition and the previous data acquisition can be displayed on the same graph.Second? when the x-coordinate data is a cyclic change in a range, I want the x-coordinate data to be displayed within the range instead of forward.
thanks for your reading.i look forward to your response. my email address is 1701213930@pku.edu.cn , feel free to contact.

**Describe the solution you'd like**
Do a display range adjustable X-axis，Add repeatable options when displaying the same contents.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mastrolinux, 2019-10-28:
There is no such serial plotter in this IDE. So I am going to close this issue.

closed 2019-10-28 by mastrolinux

breaker123, 2019-10-28:
Who should I talk to about this problem? Thank you

Closed #115.

mastrolinux, 2019-10-28:
https://github.com/arduino/arduino is the right repo.

breaker123, 2019-10-28:
thanks for your response

 
&mdash;