arduino/arduino-pro-ide #116: v0.0.2-alpha.preview - esptool rejects file path

realbuxtehuder, 2019-10-28:
[image]

When trying to flash an ESP32 board, the esptool fails with the message that it couldn't find the .bin file.
With classic IDE it works, so I guess the pro IDE passes the path to the epstool in an other format. Maybe the mix of double backslashes and slashes is the problem?

 - OS: Win10
 - Version 0.0.2-alpha.preview

mastrolinux, 2019-10-28:
@cmaglie is this in issue in the builder?

masci, 2019-10-28:
Related: https://github.com/arduino/arduino-cli/issues/163

cmaglie, 2019-10-29:
https://github.com/arduino/arduino-cli/issues/163 has been fixed, running `arduino-cli` on master should solve the problem.

per1234, 2021-02-18:
Closing as fixed by https://github.com/arduino/arduino-cli/pull/457. Thanks for reporting this @realbuxtehuder!