arduino/arduino-pro-ide #117: Tabs and files

bliskavka, 2019-10-28:
It would be great to have an ablility to close tabs. Lets say I have a few tabs and I need only two of them, others just mess around and I`d like to close them.
[image]

Also "Open Project" opion is needed in File menu, so I could have few windows with different projects, as it was in Classic version. Now it only has "Open file" option, which adds selected file to tab list.

Fede85, 2019-10-29:
What you are describing is similar to what was reported in the issue #29. It has been closed with the explanation that closing files and tabs was disabled in the classic mode to copy the behavior of the Arduino classic IDE.

I find confusing that switching mode on the same IDE you get different behavior on file management.

The Arduino classic IDE has two characteristics (not essentially features):
1. open all the files (.h, .c, cpp, other .ino, ...) contained in the sketch folder when a sketch composed by multiple files is opened. More or less like when you open a project with other "pro" IDEs. In the Arduino pro-IDE you have to manually open the files one by one.
2. allow to open only one sketch (/project) per window. A window contains only the tabs related to the project and prevents you open other files that doesn't belong to the project (/sketch folder). With the pro-IDE you can open multiple sketches in the same window.

I think for beginners the Arduino classic IDE provides an essential experience but at least very clean. If you intend to provide the same experience maybe it would nice to consider the two points above.

Although, this will break even more the consistency with the advanced mode.
This is a matter of UX design. In my opinion the two modes of the pro-IDE should be consistent with each other rather than offer two complete different experiences.

per1234, 2020-03-16:
Closing as duplicate of https://github.com/arduino/arduino-pro-ide/issues/29, https://github.com/arduino/arduino-pro-ide/issues/193, https://github.com/arduino/arduino-pro-ide/issues/133

closed 2020-03-16 by per1234