arduino/arduino-pro-ide #118: Problem on sketch rename

Drakelive, 2019-10-28:
I renamed the project name (.ino) now I have errors in the compilation. A project clean would be needed.

[image]

tattyapollo, 2020-02-14:
My guess is that since you changed the .ino name but didn't change the folder name you're running into this issue. For whatever reason the IDE, both pro and regular, really wants them to have the same name. IMO there should be a way or, if there is already one, an easier way to change both the .ino and folder name (like a "save as" or something)

per1234, 2021-02-18:
Hi @Drakelive . Thanks so much for taking the time to report this serious bug!

This should now be fixed in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4). Whether you use the "Explorer" side bar, **File > Save as...**, or the tab menu to rename the sketch, you should not get this "unable to find" error.

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234