arduino/arduino-pro-ide #119: i can't see my all ports even when i checked "show all ports"

Master811129, 2019-10-29:
**Describe the bug**
i can't see my all ports (virtual) even when i checked "show all ports"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Select Other Board & Ports'
2. tick the 'show all ports'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
classic arduino IDE:
[image]

**Desktop (please complete the following information):**
 - OS: [Win 10 enterprise]

cmaglie, 2019-10-30:
Do you have a port called "ARDUINO"? I've never seen this on windows, can you give more details on this serial port (and also on COM5 that does not appear on the port list)?

Master811129, 2019-10-30:
> Do you have a port called "ARDUINO"? I've never seen this on windows, can you give more details on this serial port (and also on COM5 that does not appear on the port list)?

ARDUINO and COM5 are virtual COM ports i use these ports to sync my programmer with Arduino IDE (i can do this with classic IDE without any problem) but this IDE couldn't detect any virtual port.
[image]

Master811129, 2019-10-31:
> Which program are you using to emulate the com port? I'd like to reproduce the bug locally

[com0com.zip](https://github.com/arduino/arduino-pro-ide/files/3792531/com0com.zip)

Master811129, 2019-11-14:
> > Which program are you using to emulate the com port? I'd like to reproduce the bug locally
> 
> [com0com.zip](https://github.com/arduino/arduino-pro-ide/files/3792531/com0com.zip)

could you see the bug?

kittaakos, 2019-11-26:
> i can't see my all ports (virtual) even when i checked "show all ports"

@Master811129, can you see the port if you list it from the [CLI](https://github.com/arduino/arduino-cli#get-the-latest-package):
```
./arduino-cli board list
Port                            Type              Board Name          FQBN            Core       
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown                                        
/dev/cu.MALS                    Serial Port       Unknown                                        
/dev/cu.SOC                     Serial Port       Unknown                                        
/dev/cu.usbmodem14401           Serial Port (USB) Arduino/Genuino Uno arduino:avr:uno arduino:avr
```

My port is `/dev/cu.usbmodem14401` for the connected board.

kittaakos, 2019-11-28:
@Master811129, I have downloaded and installed `com0com` to my Windows 10 image, can you please help with the steps setting up the virtual ports?

kittaakos, 2019-11-28:
I have set up a virtual port and mapped `COM5` to `ARDUINO`:
[image]

I can confirm, I can see the ports in the Java IDE:
[image]

And I can also confirm when I run the `board list` on the CLI, I cannot see the virtual ports. I think it is a CLI issue. What do you think @masci?

```
$ ./arduino-cli.exe board list
Port Type        Board Name FQBN Core
COM1 Serial Port Unknown
COM5 Serial Port Unknown
```

```
$ ./arduino-cli.exe board list --format json
[
  {
    "address": "COM1",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "COM5",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  }
]

```

kittaakos, 2019-12-03:
Upstream issue: https://github.com/arduino/arduino-cli/issues/505