arduino/arduino-pro-ide #126: Pin Mapping for NodeMCU unknown or incorrect

DavidBates, 2019-10-31:
**Describe the bug**
Running 0.0.2 (0.0.2.20191025.13) on OSX Catalina here and working with a NodeMCU-12E board selected. 
It seems like LED_BUILTIN works and blinks fine but the pinouts for all the 'D#' pins are either incorrect or simply not working. IE no DigitalWrite or Read function performs properly

**To Reproduce**
Steps to reproduce the behavior:
1. [Past in the linked blink sketch](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/Blink/Blink.ino)
2. Select `NodeMCU 1.0 (ESP-12E Module)` and the appropriate port in the select board screen
3. Run it should blink onboard LED
4. Modify `LED_BUILTIN` to `D2` Wire LED + Resistor to `D2` and `GND`
4. Run it, no LED light

**Expected behavior**
I would expect if D2 did not map correctly that I would be able to edit the library and/or pinout arrangement or use `GPIO04` designations like in previous versions of Arduino classic.

**Screenshots**
[image] 

**Desktop (please complete the following information):**
 - OS: OSX Catalina
 - Version Arduino Pro IDE: 0.0.2 (0.0.2.20191025.13)

**Additional context**
None

ChompsR, 2020-04-21:
Hi, I have the same problem, did you solved it?

DavidBates, 2020-06-18:
> Hi, I have the same problem, did you solved it?

I did not.

ChompsR, 2020-06-18:
David!

I did solved the problem, at first I thought that the firmware from the NodeMCU was damaged, but I kept calm, and coded a totally different program than the one I was trying to get into the device. A simple blinking led must do the trick, you must assign the pins with their names on the board, for example, instead of writing 4 you must use D2 which is the number of pin written on your board, I went pin by pin making the LED blink just to assure that everything was o.k. It is important that you leave the communication pins unwired while uploading the code, if my memory doesn’t fail me this pins are D1 and D2, Rx and Tx.

Also verify that you’re working at 9600 baud or 115200 baud, sometimes the settings are at different bauds so you have to make sure every setting is aligned.

It really is a problem of keep trying and trying this took me like 2 work days and basically the problem solved itself. I didn’t even need to upload a new firmware to the board. BUT it is important to remember that you must have installed the driver for the NodeMCU board which is called: CH340G Driver.

Greetings, friend!

DavidBates, 2020-06-19:
Interesting. this must have been resolved in a later version as I was using the names for the pins. I would run into either an unpredictable output (the labeled pin would not resolve to GPIO4. I'll test again with the latest and reopen if not resolved. Thanks.

closed 2020-06-19 by DavidBates