arduino/arduino-pro-ide #127: Serial Monitor Not working for my NodeMCU 1.0 board

techiesms, 2019-11-01:
**Describe the bug**
I just upload a normal HTTP Basic code on my NodeMCU board and when I open the serial monitor in new Arduino Pro IDE I was not getting Any response there. While I open the Serial monitor on Classic Arduino IDE I was getting the response there

**Expected behavior**
I should also get the reponse on serial monitor of Arduino Pro IDE

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 10.14.6

Master811129, 2019-11-02:
I use ESP32, same problem. Serial plotter isn't working.

kittaakos, 2019-11-26:
@techiesms, @Master811129 can you please share your `ino` file or its stripped-down version? I want to reproduce the bug locally and then come up with a possible fix. Thank you for your help!

kittaakos, 2019-11-27:
> upload a normal HTTP Basic code

Assuming you are using the `BasicHttpClient.ino` sketch from [here](https://github.com/esp8266/Arduino/blob/13f374666dfb0225167640bf695e4a943dc55b51/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino). If no, please give us an update, @techiesms. Thanks!

kittaakos, 2019-11-27:
I did the followings to reproduce the issue:
 - Download the latest Java IDE (`1.8.10`),
 - Added http://arduino.esp8266.com/stable/package_esp8266com_index.json as an additional board manager URL.
 - I have installed the `esp8266@2.6.1` core from `ESP8266 Community`.
 - Attached my `ESP8266 ESP-12E WiFi` board.
 - I selected the `NodeMCU 1.0 (ESP-12E Module)` board.
 - I verified the [`BasicHttpClient.ino`](https://raw.githubusercontent.com/esp8266/Arduino/13f374666dfb0225167640bf695e4a943dc55b51/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino) sketch.
 - I uploaded the sketch, it failed.

The error was:
```
Arduino: 1.8.10 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:
IROM   : 254312          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 26688   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1244  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1868  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25040 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 284112 bytes (27%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28152 bytes (34%) of dynamic memory, leaving 53768 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Traceback (most recent call last):
Serial port /dev/cu.usbserial-0001
  File "/Users/akos.kitta/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/upload.py", line 69, in <module>
    esptool.main(cmdline)
  File "/Users/akos.kitta/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/esptool/esptool.py", line 2889, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/Users/akos.kitta/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/esptool/esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/akos.kitta/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/pyserial/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Users/akos.kitta/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/pyserial/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/Users/akos.kitta/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/pyserial/serial/serialposix.py", line 438, in _reconfigure_port
    [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
termios.error: (22, 'Invalid argument')
termios.error: (22, 'Invalid argument')

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Please provide the steps you did. Otherwise, I cannot reproduce and fix the defect. Thank you!

locotron, 2019-12-09:
Same happening with a esp8266 using the NodeMCU 1.0 ESP-12E Module.
Serial monitor on the Arduino Pro ide doesn't show up anything. Tried with normal and advanced mode to see if it made any difference, results are the same.
Normal arduino ide shows the data on the serial monitor.

kittaakos, 2019-12-09:
@locotron, can you please share your sketch that works with the Java-based Arduino IDE but not the PRO IDE?

locotron, 2019-12-09:
@kittaakos 
`#include <NTPClient.h>
// change next line to use with another board/shield
#include <ESP8266WiFi.h>
//#include <WiFi.h> // for WiFi shield
//#include <WiFi101.h> // for WiFi 101 shield or MKR1000
#include <WiFiUdp.h>

const char *ssid     = "<SSID>";
const char *password = "<PASSWORD>";

WiFiUDP ntpUDP;

// By default 'pool.ntp.org' is used with 60 seconds update interval and
// no offset
NTPClient timeClient(ntpUDP);

// You can specify the time server pool and the offset, (in seconds)
// additionaly you can specify the update interval (in milliseconds).
// NTPClient timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);

void setup(){
  Serial.begin(115200);
  WiFi.begin(ssid, password);

  while ( WiFi.status() != WL_CONNECTED ) {
    delay ( 500 );
    Serial.print ( "." );
  }

  timeClient.begin();
}

void loop() {
  timeClient.update();

  Serial.println(timeClient.getFormattedTime());

  delay(1000);
}`

Same used in https://github.com/arduino-libraries/NTPClient example.

kittaakos, 2019-12-09:
For the record, on how to set up basic HTTP client/server over WiFi for ESP8266:
 - 3rd party board: `http://arduino.esp8266.com/stable/package_esp8266com_index.json `
 - Install `esp8266@2.6.2` core from `ESP8266 Community`.
 - `Sketch` > `Include Library` > `ESP8266WiFi`

Code:
```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#ifndef STASSID
#define STASSID "YOUR_SSID"
#define STAPSK  "YOUR_PWD"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

const char* host = "ipv4 - Wi-Fi is connected to "YOUR_SSID" and has the IP address and start a server `nc -v -l 3000`";
const uint16_t port = 3000;

ESP8266WiFiMulti WiFiMulti;

void setup() {
  Serial.begin(115200);

  // We start by connecting to a WiFi network
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid, password);

  Serial.println();
  Serial.println();
  Serial.print("Wait for WiFi... ");

  while (WiFiMulti.run() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  delay(500);
}

void loop() {
  Serial.print("connecting to ");
  Serial.print(host);
  Serial.print(':');
  Serial.println(port);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;

  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    Serial.println("wait 5 sec...");
    delay(5000);
    return;
  }

  // This will send the request to the server
  client.println("hello from ESP8266");

  //read back one line from server
  Serial.println("receiving from remote server");
  String line = client.readStringUntil('\r');
  Serial.println(line);

  Serial.println("closing connection");
  client.stop();

  Serial.println("wait 5 sec...");
  delay(5000);
}
```

Serial-monitor output:
```
14:41:09.588 -> Wait for WiFi... .....
14:41:13.280 -> WiFi connected
14:41:13.280 -> IP address: 
14:41:13.280 -> 192.168.0.104
14:41:13.800 -> connecting to 192.168.0.103:3000
14:41:13.906 -> receiving from remote server
14:41:18.899 -> 
14:41:18.899 -> closing connection
14:41:18.899 -> wait 5 sec...
14:41:23.914 -> connecting to 192.168.0.103:3000
14:41:23.914 -> connection failed
14:41:23.914 -> wait 5 sec...
14:41:28.923 -> connecting to 192.168.0.103:3000
14:41:28.923 -> receiving from remote server
14:41:18.899 -> 
14:41:18.899 -> closing connection
14:41:18.899 -> wait 5 sec...
14:41:23.914 -> connecting to 192.168.0.103:3000
14:41:23.914 -> connection failed
14:41:23.914 -> wait 5 sec...
...
```

Terminal output:
```
% nc -v -l 3000
hello from ESP8266
% nc -v -l 3000
hello from ESP8266
...
```

kittaakos, 2019-12-09:
When I connect my ESP8266 board, the CLI does not know its name (`arduino-cli board list --format json`), although the corresponding ESP8266 core is installed. Maybe that is the root cause of the defect... I am on it.

```
% ./arduino-cli board list --format json
[
  {
    "address": "/dev/cu.SOC",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.MALS",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.Bluetooth-Incoming-Port",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.usbserial-0001",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)"
  }
]%                            
```

kittaakos, 2019-12-09:
> Maybe that is the root cause of the defect

Yes, that was the issue. Thank you for your help, @locotron. 🏅

The bug was fixed as part of #112. The fix will be available with the next release.

locotron, 2019-12-09:
> When I connect my ESP8266 board, the CLI does not know its name (`arduino-cli board list --format json`), although the corresponding ESP8266 core is installed. Maybe that is the root cause of the [...]
In ca

> > Maybe that is the root cause of the defect
> 
> Yes, that was the issue. Thank you for your help, @locotron. 🏅
> 
> The bug was fixed as part of #112. The fix will be available with the next release.

Well done!

closed 2019-12-11 by kittaakos

ubidefeo, 2020-10-17:
@akshaycool040506 
this could be an issue with serial syncing, as in the board and the serial monitor fail to start communicating at the same time/speed.
Please make sure you set the right speed for both, and if you could test using Arduino IDE 1.8.13 and let us know if you have the same issue that would help us a lot to look for where the problem is.

Thank you
u.