arduino/arduino-pro-ide #129: Cannot type parentheses on number 9 key

kanokkorn, 2019-11-02:
**Describe the bug**
Cannot type parentheses on number 9 key

**To Reproduce**
Steps to reproduce the behavior:
1. press parentheses on number 9

**Expected behavior**
expected a " ( " 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: 10.0.18362 Build 18362

satiowadahc, 2019-11-06:
Are you using a US keyboard layout? If you do a command line `systeminfo` what's the output of System locale and Input locale?

kanokkorn, 2019-11-07:
> Are you using a US keyboard layout? If you do a command line `systeminfo` what's the output of System locale and Input locale?

Yes, I'm using a US keyboard layout. Here is the output.
```
System Locale:             en-us;English (United States)
Input Locale:              en-us;English (United States)
```

OriginalAdric, 2019-11-10:
Trying to clarify the issue: The 3 and 9 keys are being handled as PgDn and PgUp, respectively, both on the number row and numpad. The numpad behavior occurs regardless of Numlock status. The only difference in behavior I've observed is that Shift+3 and Shift+9 also select any text the cursor passes while Shift+Numpad3 and Shift+Numpad9 do not.

This occurs on my desktop Windows PC with full keyboard, both with physical keyboard and on-screen keyboard.
This **does not** occur on my Surface Pro 3 with magnetic keyboard, either with physical or on-screen keyboards.

Sheel-1609, 2019-11-15:
Hi, i have the same problem on my HP laptop. @OriginalAdric is right this 9 and 3 are acting as page up and page down respectively. I have tried this with both US and UK keyboard styles the same problem persists. same with the numpad numbers. This problem also exists with an external keyboard.

flaviosg, 2019-11-15:
I have the same problem. I'm using US English International Keyboad layout, with Windows 10. 
"9" key acts as PgUp and "3" key acts as PgDw. 
I can only type "3" using ALT+51 and "9" using ALT+57

Ometeo, 2019-11-21:
Hi, I have the same problem with the French Keyboard layout, if it helps.
OS : Windows 10
Keyboard : Razer Anansi

jankeromnes, 2019-11-29:
I wasn't able to reproduce this bug in Arduino Pro v0.0.3, on both Windows 7 and Mac OS:

- inputting 3s and 9s always worked fine (from on-screen keyboard, physical keyboard, numrow and numpad)
- the only way I could get a 3 or 9 to act as different keys was by turning off NumLock and using the numpad (numrow and on-screen keyboard still input 3s and 9s fine without NumLock)

Maybe this bug is specific to Windows 10?

kittaakos, 2019-12-09:
I could reproduce it neither on Windows 10 nor macOS; worked as expected with EN_US layout. I am going to try it again with a real Windows machine with a numpad.

kittaakos, 2019-12-09:
It also works on Windows 7 with numpad. I have tried it with the physical and the on-screen keyboard too.

spoenemann, 2020-01-10:
I cannot reproduce this on my Windows 10 machine.

@kanokkorn @OriginalAdric @Sheel-1609 @flaviosg @Ometeo
Could you help analyzing the problem?

1. https://w3c.github.io/uievents/tools/key-event-viewer.html
2. Enter `39` in the input field
3. Make a screenshot of the web page and attach here (if it's the same output as already posted by others, there's no need to post it again)

kanokkorn, 2020-01-10:
I think I can reproduce this by 
1. changing the language to non-US layout
1. type with a non-US layout
1. switch back to US layout

Test on Windows 10 and Ubuntu 18.04 TLS but can't reproduce this bug on  Ubuntu. I will test again on a different machine.

spoenemann, 2020-01-13:
Finally I could reproduce this. Cause: the mapping returned by [native-keymap](https://www.npmjs.com/package/native-keymap) contains an entry

```json
    "Numpad3": {
        "vkey": "VK_NEXT"
    }
```

`VK_NEXT` is the Windows virtual key code for _PageDown_; this leads to a keybinding mapping where _PageDown_ is replaced with _3_.

spoenemann, 2020-01-13:
This must be fixed in Theia:
https://github.com/eclipse-theia/theia/issues/6872

spoenemann, 2020-02-03:
Should be ok with the next update.

closed 2020-02-03 by spoenemann