arduino/arduino-pro-ide #132: Serial monitor struggles to communicate

RunarSJ, 2019-11-06:
I uploaded a serial passthrough code to a an arduino uno using the softwareserial library to communicate with a hm10 bluetooth module. Configuring did not work at all in PRO IDE environment, but worked instantly in the normal IDE environment.

kittaakos, 2019-11-26:
> uploaded a serial passthrough code to a an arduino uno using the softwareserial library to communicate with a hm10 bluetooth module

> but worked instantly in the normal IDE environment

@RunarSJ, can you please me give some hints (screenshot, screencast, or whatever is the easiest for you) on reproducing this? What exactly did you do in the Java IDE and the PRO IDE? What board did you use, what cores were installed, and so on? Thank you!

mastrolinux, 2020-01-08:
Closing due to lack of info.

closed 2020-01-08 by mastrolinux