arduino/arduino-pro-ide #133: Add close sketch option

sevsev9, 2019-11-07:
**Is your feature request related to a problem? Please describe.**
There is no proper way to close a sketch or any other window really in the IDE. So it just keeps filling up the window bar until you close the IDE.

**Describe the solution you'd like**
A small "x" on every tab open and the CTRL+W shortcut to close focused tabs fast.

**Describe alternatives you've considered**
Maby also add to context menu of the tabs' header.

svenefftinge, 2019-11-28:
It is intentionally omitted in the default classic mode, as it is meant to be used on a single sketch that has all its files open. You can switch to advanced mode, where the close option is present (the cross in the tab, see screenshot).

[image]

sevsev9, 2019-11-28:
In this case I would recommend though that you would automatically open a sketch(project) in a new window so that it is not as confusing.

closed 2019-11-28 by sevsev9