arduino/arduino-pro-ide #134: arduino:avr:uno is not installed +  failed to listen: listen tcp :50051: bind

fhtino, 2019-11-10:
Arduino Pro IDE ( 0.0.2 )shows this error at startup:
"Editor support may be inaccurate because the core for the board arduino:avr:uno is not installed. Use the Boards Manager to install it."

Running from command line, I see this error when the IDE tries to start arduino-cli:
```
root INFO Error: Command failed: "C:\mybin\arduino-pro-ide-0.0.2-alpha.preview-windows\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" daemon -v --log-level info --format json --log-format json
2019/11/10 16:34:29 failed to listen: listen tcp :50051: bind: An attempt was made to access a socket in a way forbidden by its access permissions.
    at ChildProcess.exithandler (child_process.js:299:12)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Socket.stream.socket.on (internal/child_process.js:381:11)
    at Socket.emit (events.js:182:13)
    at Pipe._handle.close (net.js:606:12)
daemon INFO <<< The 'arduino-cli' daemon is up an running.
```

Running ProIDE or directly Arduino-cli as administrator does not solve the issue.

OS: Windows 10 -1809

cc /@akash73

kittaakos, 2019-11-13:
@fhtino, can you please try to extract and run the application somewhere under `%USERPROFILE%`?

fhtino, 2019-11-13:
@kittaakos I've run it under C:\Users\Fabry\temp\arduino-pro-ide-0.0.2-alpha.preview-windows\ 
Nothing changed: same errors on IDE and from command-line.

fhtino, 2019-11-13:
@kittaakos I've found a similar issue here https://github.com/docker/for-win/issues/3171
I have HyperV installed. I've tried netsh on my PC. Port 50051 is included in the exclusion ranges. 

```
C:\>netsh interface ipv4 show excludedportrange protocol=tcp
Protocol tcp Port Exclusion Ranges
Start Port    End Port
----------    --------
      5357        5357
      5700        5700
      8884        8884
     49863       49962
     50042       50141
     64380       64479
     64746       64845
* - Administered port exclusions.
```

kittaakos, 2019-11-14:
Thank you for your help, @fhtino. The CLI part is not under my control 😊, but I have opened a [proposal](https://github.com/arduino/arduino-cli/issues/485). Hopefully, Windows users with Hyper-V installed, do not have to do any magic to be able to use the IDE.

If you have a better idea, please let us know.

kittaakos, 2020-04-08:
This will be available with the 0.0.6-alpha release. You can use the CLI config to update the `daemon.port`.

I am moving this item to the current milestone as this feature will be supported by the next release.

kittaakos, 2020-05-05:
@fhtino, you should be able to tune the daemon port via the CLI configuration.
`File` > `Settings` > `Open CLI Configuration` and change the `daemon.port` entry.

If you have questions/problems, please let us know.

closed 2020-05-05 by kittaakos