arduino/arduino-pro-ide #135: The layout of the main toolbar items (toggle pro-mode, serial monitor) is broken

kittaakos, 2019-11-12:
**Describe the bug**
Refer to the bug title and see the attached screencast:
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] all
 - Version [e.g. 22] from source, `HEAD`.

**Additional context**
Add any other context about the problem here.

jbicker, 2020-01-15:
There is space reserved for the labels which are shown on hovering a button. That's why the toolbar does not get smaller. So it is not really broken.
Any suggestions how we should proceed with that?
[image]

per1234, 2021-02-18:
I think this no longer occurs due to the provisional workaround described in https://github.com/arduino/arduino-pro-ide/issues/337#issuecomment-688359549, but will probably come back if a try fix is found for that and the minimum window size is made more reasonable.