arduino/arduino-pro-ide #140: Unable to load code for ESP32 Dev Module board

dnsgusttavo, 2019-11-20:
**Describe the bug**
Unable to load code for ESP32 Dev Module board

**OUTPUT ERROR**
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'c:\\Users\\DNSGS\\Documents\\Arduino\\sketch_oct12a\\sketch_november20a/sketch_november20a.esp32.esp32.esp32.partitions.bin'
Upload error: Error: uploading error: exit status 2

bastian2001, 2020-01-12:
I have the same problem with another error. I have also tried manually putting the ESP32 into download mode. Same error with or without doing that:

Traceback (most recent call last):
  File "esptool.py", line 2959, in <module>
  File "esptool.py", line 2952, in _main
  File "esptool.py", line 2652, in main
  File "esptool.py", line 222, in __init__
  File "site-packages\serial\__init__.py", line 88, in serial_for_url
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM7': WindowsError(5, 'Zugriff verweigert')
Failed to execute script esptool
Upload error: Error: 2 UNKNOWN: uploading error: exit status 4294967295

Windows 10 1903 on a Ryzen 7 1700 with Arduino Pro IDE 0.0.3-alpha.preview

per1234, 2021-02-18:
Hi @gustavomaltez. Thanks so much for taking the time to report this serious bug! It turned out to be a problem with the Arduino CLI tool the Arduino Pro IDE uses under the hood for a lot of the Arduino-specific things that aren't related to the GUI. It was fixed by https://github.com/arduino/arduino-cli/pull/457

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234