arduino/arduino-pro-ide #142: Proposal for changing the way we select and use the CLI inside the IDE

kittaakos, 2019-11-20:
**Currently behavior**:
The IDE picks up and uses the `arduino-cli` from the `PATH` if present instead of the bundled one.

**Problem**:
We do check whether the `arduino-cli` from the `PATH` is compatible with the IDE. It could lead to strange behavior, such as one described [here](https://github.com/arduino/arduino-pro-ide/issues/21#issuecomment-553469941).

**Expected behavior**:
 - If `arduino-cli` is not on the `PATH`, use the embedded one.
 - If `arduino-cli` is on the `PATH` apply the following rules:

   - Compare the versions of the embedded one and the one from the `PATH`.
   - Use the higher version.
   - If the versions match, use the embedded one.  (Note, we **do not** check the commit hash in the versions when comparing them.)
   - Always log which one was selected at app startup.
   - The log message must contain the commit hash, although we do not use it when comparing the versions.

masci, 2019-11-21:
From my personal experience on similar software, the IDE should always use the embedded version of the CLI, which should be considered as part of the IDE release package. The only exception might be an "advanced setting" where you can override the embedded binary if you know what you're doing.

kittaakos, 2019-11-21:
Thank you for your feedback, @masci.

> From my personal experience on similar software, the IDE should always use the embedded version of the CLI, which should be considered as part of the IDE release package

We had the same thought during our call yesterday. The current limitation: we do not support updates inside the app, so there is no way for users trying out a more recent version of the CLI without downloading and installing a new version of the app.

> The only exception might be an "advanced setting"

Unfortunately, the preferences support still on a branch, hence we decided with this logic. Once we merge the preferences, we should adjust the _Expected behavior_ too and do exactly what you [wrote above](https://github.com/arduino/arduino-pro-ide/issues/142#issuecomment-557005958). 👆

Thoughts?

masci, 2019-11-21:
> The current limitation: we do not support updates inside the app, so there is no way for users trying out a more recent version of the CLI 

I understand this is something valuable at alpha/beta stages, take my concerns as only relevant in the long run, when having the IDE to support multiple versions of the CLI will be incredibly hard and will take a toll in both the IDE and the CLI release process.

> Unfortunately, the preferences support still on a branch

Depending on how user friendly we want to be, have you considered relying on a special env var like `ARDUINO_CLI_PATH` that defaults to the embedded binary when empty/not defined? This way you would support CLI updates that are independent from the IDE without introducing a behaviour (the "auto-detection" mechanism) that might be hard to get rid of later. But again, this would make user documentation a bit trickier if compared to "just update the CLI, the IDE will pick it up".

closed 2020-08-27 by per1234