arduino/arduino-pro-ide #146: No "flavours" section on Select Board dialog

pirropirro, 2019-11-22:
**Describe the bug**
Inside the `Select Board` dialog, when I choose a board (es. Arduino Mini) I would expect to see the flavours section in order to change boards' custom options. 

**To Reproduce**
- Open the `Select Board` dialog
- Choose a customizable board (es. `Arduino Mini`) 

**Expected behavior**
The flavours section should appear with the selected board options

**Screenshots**
### **Behavior on the `Arduino Web Editor`**

[image]

<hr />

### **Behavior on  the `Arduino Pro IDE`**

[image]

**Desktop (please complete the following information):**
 - OS: any
 - Version any

kittaakos, 2019-11-22:
> The flavours section should appear with the selected board options

What is `flavours`, can you please specify it?  Does the CLI capable of providing this info?

pirropirro, 2019-11-22:
Some board (es. 'Arduino Mini') are distributed with the same fqbn (`arduino:avr:mini`) but with some different specifications (as the processor, CPU frequency, etc...) defined as extra fields in the fqbn (`arduino:avr:mini:cpu=atmega168;`).

In the Web IDE (as you can see in the second screenshot) these kinds of differences are defined as 'flavours' and the user can choose the right board.

Fortunately, the CLI has a command (`boardDetails`) where you can retrieve these options but the IDE still not handle them.

kittaakos, 2019-11-22:
👍 Cool, thanks for explaining! Are you working on this item? Just asking why is it scheduled for the milestone? Did you put it in the milestone on purpose?

pirropirro, 2019-11-22:
Yup, I'm working on it (I forgot to assign it to myself :sweat_smile:) but I just put it on the backlog and not in the milestone scope.

kittaakos, 2020-04-09:
Same as #10.

closed 2020-05-05 by kittaakos