arduino/arduino-pro-ide #147: Theme and managers resetted when switching mode

GoodLuck95, 2019-11-22:
If dark theme is active, and mode is switched (either to advanced or to classic one), the light theme is turned back on, instead of keeping the dark one.

Same behavior for library and board managers, that, if opened in the left panel, they are closed when changing mode.

kittaakos, 2019-11-22:
Thanks for reporting, @GoodLuck95!

> If dark theme is active, and mode is switched (either to advanced or to classic one), the light theme is turned back on, instead of keeping the dark one.

This has been fixed and will be available with the next release.

> Same behavior for library and board managers, that, if opened in the left panel, they are closed when changing mode.

My understanding was that this is the desired behavior, is it? (CC: @mastrolinux)

GoodLuck95, 2019-11-22:
Yes, at least in my opinion, it is the desired behaviour

kittaakos, 2019-12-11:
> This has been fixed and will be available with the next release.

Was fixed in `0.0.3`.

closed 2019-12-11 by kittaakos