arduino/arduino-pro-ide #150: Variable not marked

Alexagomesmar65, 2019-11-25:
String type variable is not marked in Dark mode as variable.

Recognizes the release as a variable, but does not mark as it does with "char" or "int" types.

svenefftinge, 2019-11-28:
Can you elaborate aa bit more on this or share a screenshot. I'm not entirely sure what you mean.

Master811129, 2020-01-13:
i think he's right true the String class is not colored in dark theme (and also Arduino theme for me)
[image]
[image]

per1234, 2021-02-17:
Closing as duplicate of https://github.com/arduino/arduino-pro-ide/issues/285. Even though this one came first, the mention of the  theme could lead to confusion. As Master811129 kindly pointed out, this issue occurs regardless of which theme you're using.

closed 2021-02-17 by per1234