arduino/arduino-pro-ide #151: Function Bar - option Selection

Alexagomesmar65, 2019-11-25:
In Windows 10, my arduino.Pro.IDE version 0.0.2 does not show the "Selection" option in the Function bar. I know of the existence, by video on youtube. https://www.youtube.com/watch?v=VToBlkgJKJE

kittaakos, 2019-12-05:
Thank you for reporting it, @Alexagomesmar65.
Unfortunately, it is a known issue in Theia: https://github.com/eclipse-theia/theia/issues/6425

per1234, 2021-02-18:
This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

Make sure to set the Pro IDE to "Advanced Mode" by moving the switch on the right side of the function bar to the right hand position. The "Selection" menu is intentionally hidden when in "Classic" mode.

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234