arduino/arduino-pro-ide #154: Language Support stops working

svenefftinge, 2019-11-28:
Opening an .ino file and then doing some edits like e.g. adding the following line:
```
const char* string = "foo%test";
```
eventually results in:
`jsonrpc2: connection is closed``
after which nothing works anymore.

Workaround: Run `View: Reset Workbench Layout`.

**Desktop (please complete the following information):**
 - OS: OSX
 - Version v0.0.3

jankeromnes, 2019-12-11:
I wasn't able to reproduce this when I tried, but after a while this just happened to me randomly (browser app).

I simply selected board "Arduino Zero", then installed the core, then selected board "Arduino Zero (programming port)", then tried building the sketch.

The `jsonrpc2: connection is closed` message started appearing in the logs.

<details>
<summary>(view full logs)</summary>

```
$ theia start
root INFO Theia app listening on http://localhost:3000.
daemon INFO >>> Starting arduino-cli version: 0.6.0 commit: 2049a7a daemon from /workspace/arduino-editor/arduino-ide-extension/build/arduino-cli...
daemon INFO Checking for config file in: /home/gitpod/.arduino15/arduino-cli.yaml
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Candidate IDE Directory: /workspace/arduino-editor/arduino-ide-extension/build
daemon INFO CLI is not bundled into the IDE
daemon INFO Checking for config in: /arduino-cli.yaml
daemon INFO error loading [error: open /arduino-cli.yaml: no such file or directory]
daemon INFO Checking for config in: /workspace/arduino-cli.yaml
daemon INFO error loading [error: open /workspace/arduino-cli.yaml: no such file or directory]
daemon INFO Checking for config in: /workspace/arduino-editor/arduino-cli.yaml
daemon INFO error loading [error: open /workspace/arduino-editor/arduino-cli.yaml: no such file or directory]
daemon INFO Checking for config in: /workspace/arduino-editor/browser-app/arduino-cli.yaml
daemon INFO error loading [error: open /workspace/arduino-editor/browser-app/arduino-cli.yaml: no such file or directory]
daemon INFO Configuration set
daemon INFO arduino-cli-0.6.0
daemon INFO Starting root command preparation (`arduino`)
daemon INFO Formatter set
daemon INFO <<< The 'arduino-cli' daemon is up an running.
root INFO Using Git [2.20.1] from the PATH. (/usr/bin/git)
root INFO Starting language server /workspace/arduino-editor/arduino-ide-extension/build/arduino-language-server -clangd /usr/bin/clangd -cli /workspace/arduino-editor/arduino-ide-extension/build/arduino-cli
root INFO [nsfw-watcher: 5407] Started watching: /home/gitpod/.theia
root INFO [nsfw-watcher: 5407] Started watching: /home/gitpod/Arduino/sketch_december11a
Started watching: /home/gitpod/Arduino/sketch_december11b
Started watching: /home/gitpod/Arduino/sketch_december11a
Started watching: /home/gitpod/Arduino/sketch_december11b
Started watching: /home/gitpod/Arduino/sketch_december11a/sketch_december11a.ino
root ERROR Arduino: 2019/12/11 12:33:37 Command error: /workspace/arduino-editor/arduino-ide-extension/build/arduino-cli exit status 1
2019/12/11 12:33:37 ------------------------------BEGIN STDERR
 Error during build: platform not installed
 ------------------------------END STDERR

root INFO  >>> Creating and caching a new client for file:///home/gitpod/Arduino/sketch_december11a...
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Candidate IDE Directory: /workspace/arduino-editor/arduino-ide-extension/build
daemon INFO Loading hardware from: /home/gitpod/.arduino15/packages
daemon INFO Loading package builtin from: /home/gitpod/.arduino15/packages/builtin
daemon INFO Checking existence of 'tools' path: /home/gitpod/.arduino15/packages/builtin/tools
daemon INFO Loading tools from dir: /home/gitpod/.arduino15/packages/builtin/tools
daemon INFO Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO Loaded tool [tool: Version: 1.0.0]
daemon INFO Adding libraries dir [dir: /home/gitpod/Arduino/libraries, location: sketchbook]
daemon INFO instance:<id:1 >
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
daemon INFO Loading hardware from: /home/gitpod/.arduino15/packages
daemon INFO Loading package builtin from: /home/gitpod/.arduino15/packages/builtin
daemon INFO Checking existence of 'tools' path: /home/gitpod/.arduino15/packages/builtin/tools
daemon INFO Loading tools from dir: /home/gitpod/.arduino15/packages/builtin/tools
daemon INFO Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO Loaded tool [tool: Version: 1.0.0]
daemon INFO Adding libraries dir [dir: /home/gitpod/Arduino/libraries, location: sketchbook]
daemon INFO Updating libraries index
daemon INFO Loading hardware from: /home/gitpod/.arduino15/packages
daemon INFO Loading package builtin from: /home/gitpod/.arduino15/packages/builtin
daemon INFO Checking existence of 'tools' path: /home/gitpod/.arduino15/packages/builtin/tools
daemon INFO Loading tools from dir: /home/gitpod/.arduino15/packages/builtin/tools
daemon INFO Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO Loaded tool [tool: Version: 1.0.0]
[... 147 lines cut ...]
root ERROR Arduino: 2019/12/11 12:35:21 jsonrpc2 handler: notification "textDocument/didSave" handling error: jsonrpc2: connection is closed

root ERROR Arduino: 2019/12/11 12:35:21 jsonrpc2 handler: notification "workspace/didChangeWatchedFiles" handling error: jsonrpc2: connection is closed

root ERROR Arduino: 2019/12/11 12:35:21 jsonrpc2 handler: notification "workspace/didChangeWatchedFiles" handling error: jsonrpc2: connection is closed

root ERROR Arduino: 2019/12/11 12:35:22 jsonrpc2 handler: notification "textDocument/didChange" handling error: jsonrpc2: connection is closed
```

</details>

This is what the process tree now looks like (notice the `clangd` zombie):

```
[... cut ...]
   5297    5407    5263     869 pts/0       5263 Sl+  33333   0:00  |                   \_ /home/gitpod/.nvm/versions/node/v10.16.3/bin/node /workspace/arduino-editor/node_modules/@theia/core/lib/node/messaging/ipc-bootstrap.js
   5297    6066    5263     869 pts/0       5263 Sl+  33333   0:00  |                   \_ /workspace/arduino-editor/arduino-ide-extension/build/arduino-language-server -clangd /usr/bin/clangd -cli /workspace/arduino-editor/arduino-ide-extension/build/arduino-cli -fqbn arduino:samd:arduino_zero_edbg -board-name "Arduino Zero (Programming Port)"
   6066    6071    5263     869 pts/0       5263 Z+   33333   0:00  |                       \_ [clangd.main] <defunct>
```

jankeromnes, 2019-12-12:
Oh, it sounds like Theia is already supposed to auto-restart `clangd`: https://github.com/eclipse-theia/theia/issues/6308

Could this possibly be the same bug?

ubidefeo, 2021-02-19:
This should now be working in the new 0.1.4 release of Arduino Pro IDE since we have reworked the LS entirely
This issue will be closed

closed 2021-02-19 by ubidefeo