arduino/arduino-pro-ide #156: Arduino CLI dies after multiple compilations

endorama, 2019-11-29:
**Describe the bug**
Compilation/upload is not working anymore

**To Reproduce**
Steps to reproduce the behavior:
1. Compile
1. compile again quickly on same target

**Please note** I'm not sure this will reliably reproduce the issue.

**Expected behavior**
Verify/Upload buttons continue to work.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Doing a workshop using Pro Editor as the default IDE.

I'm using an Arduino UNO, compiling Arduino code using Adafruit DotStar and FTDebouncer libraries.

The `arduino-cli` process is no longer alive.

Verify/Upload button do not work anymore: you can click on them but the compilation never starts.

Restarting the Pro Editor restore a working situation.

This is an extract of the logs:
```
root INFO upload { uri: 'file:///home/etenani/Arduino/pixelGame/pixelGame.ino',
  board:
   { name: 'Arduino/Genuino Uno',
     fqbn: 'arduino:avr:uno',
     port: '/dev/ttyACM1' },
  port: '/dev/ttyACM1' }
daemon INFO Searching tools required for board arduino:avr:uno
daemon INFO Required tool [tool: {ToolName:"arduinoOTA",ToolVersion:"1.2.1",ToolPackager:"arduino"}]
daemon INFO Required tool [tool: {ToolName:"avr-gcc",ToolVersion:"5.4.0-atmel3.6.1-arduino2",ToolPackager:"arduino"}]
daemon INFO Required tool [tool: {ToolName:"avrdude",ToolVersion:"6.3.0-arduino14",ToolPackager:"arduino"}]
daemon INFO Tool required for upload [tool: Version: 1.2.1 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
daemon INFO Tool required for upload [tool: Version: 5.4.0-atmel3.6.1-arduino2 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
daemon INFO Tool required for upload [tool: Version: 6.3.0-arduino14 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
daemon INFO Tool required for upload [tool: Version: 1.0.0 Systems: [OS: i686-pc-linux-gnu, OS: x86_64-pc-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: arm-linux-gnueabihf, OS: arm64-linux-gnueabihf]]
daemon INFO Tool required for upload [tool: Version: 5.8-arduino11]
daemon INFO Tool required for upload [tool: Version: 0.10.0-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
daemon INFO Tool required for upload [tool: Version: 7-2017q4 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
daemon INFO Tool required for upload [tool: Version: 1.7.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf]]
daemon INFO Tool required for upload [tool: Version: 4.5.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
daemon INFO Tool required for upload [tool: Version: 1.2.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
daemon INFO Daemon exited with code: null. Signal was: SIGTERM.
root INFO RangeError [ERR_CHILD_PROCESS_STDIO_MAXBUFFER]: stdout maxBuffer length exceeded
    at Socket.onChildStdout (child_process.js:353:14)
    at Socket.emit (events.js:187:15)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:260:13)
    at Socket.Readable.push (_stream_readable.js:219:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:94:17)
root ERROR Uncaught Exception:  Error
root ERROR Error
    at Http2CallStream.call.on (/home/etenani/Downloads/Arduino Pro IDE-v0.0.3-linux/resources/app/node_modules/@grpc/grpc-js/build/src/client.js:101:45)
    at Http2CallStream.emit (events.js:187:15)
    at process.nextTick (/home/etenani/Downloads/Arduino Pro IDE-v0.0.3-linux/resources/app/node_modules/@grpc/grpc-js/build/src/call-stream.js:71:22)
    at process._tickCallback (internal/process/next_tick.js:61:11)
root ERROR Request upload failed with error:  Error
    at Http2CallStream.call.on (/home/etenani/Downloads/Arduino Pro IDE-v0.0.3-linux/resources/app/node_modules/@grpc/grpc-js/build/src/call.js:68:41)
    at Http2CallStream.emit (events.js:187:15)
    at process.nextTick (/home/etenani/Downloads/Arduino Pro IDE-v0.0.3-linux/resources/app/node_modules/@grpc/grpc-js/build/src/call-stream.js:71:22)
    at process._tickCallback (internal/process/next_tick.js:61:11)
```

mastrolinux, 2019-12-23:
This should be fixed in the last version, can you please verify?

mastrolinux, 2020-01-07:
Fixed in 0.7.1

closed 2020-01-07 by mastrolinux