arduino/arduino-pro-ide #157: Serial Monitor breaks after recieveing lots of data

endorama, 2019-11-29:
**Describe the bug**
When using the serial monitor, if the amount of data is consistent and protracted in time, an error freeze it preventing any more data to be shown on the Pro Editor.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a sketch sending data to the Serial rapidly (like 20 chars with 1ms delay)
2. Upload on board (I was using an Arduino Uno)
3. Open Serial monitor
4. Wait till freeze

**Expected behavior**
Serial Monitor keeps displaying incoming data

**Desktop (please complete the following information):**
 - OS: elementary OS Juno (based upon Ubuntu 18.04)
 - Version 0.0.3

**Additional context**
Add any other context about the problem here.

Relevant logs:
```
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: yer
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b:  A s
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: core
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: : 19
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: 
pl
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: ayer
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b:  B s
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: core
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: : 1
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: 
pla
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: yer 
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: A sc
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: ore:
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b:  19
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: 
pla
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: yer 
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: B s
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: core
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: : 1
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: 
pla
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: yer 
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: A sc
root INFO Received data from 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b: ore:

monitor-service ERROR Error occurred for connection 42c0ce2c-761a-47f4-9eb7-3074cf7acd3b. Error: interrupted system call
    at Http2CallStream.call.on (/home/etenani/Downloads/Arduino Pro IDE-v0.0.3-linux/resources/app/node_modules/@grpc/grpc-js/build/src/call.js:68:41)
    at Http2CallStream.emit (events.js:187:15)
    at process.nextTick (/home/etenani/Downloads/Arduino Pro IDE-v0.0.3-linux/resources/app/node_modules/@grpc/grpc-js/build/src/call-stream.js:71:22)
    at process._tickCallback (internal/process/next_tick.js:61:11)

```

kittaakos, 2019-12-03:
@endorama, thanks for reporting. It is not about the hiuge amount of data, but this is a bug in the CLI: arduino/arduino-cli#504

kittaakos, 2019-12-09:
It will be fixed as part of #112.

I have moved this issue to the `0.0.4` milestone.

kittaakos, 2019-12-11:
Fixed, and we managed to roll it out in the `0.0.3`

closed 2019-12-11 by kittaakos