arduino/arduino-pro-ide #158: --->WHAT HAPPEN WITH YOU?

agratx, 2019-11-30:
I don't like it, because I can't use it as an ISP programmer--->WHAT HAPPEN WITH YOU?

In normal mode, it should be functional as the normal IDE, and in advanced mode as it should behave as such. I don't think you have made the great revolution that you are announcing if the transition is from earth to heaven if you have excluded all of the above in this new Arduino Pro IDE

--->WHAT HAPPEN WITH YOU?

tuxedo0801, 2020-10-01:
@agratx What happen to YOU? This is an early stage version of the IDE and not a thing where many developer years went into. Cool down and relax.

ubidefeo, 2020-10-01:
thank you @tuxedo0801 for chiming in
the issue is very old, you can ignore stuff that is more than 6-8 months, as a lot has changed in the recent months.
By the way we now support more than ISP, a whole bunch of programmers are defined by platform implementers and they mostly work :)

I appreciate you going through the issues, means you're liking the project 👍 
(Take some comments with a grain of salt, many users find it easier to throw words at you before thinking)

tuxedo0801, 2020-10-01:
@ubidefeo Then why don't you close that old stuff? For others this looks like the team is doing nothing :-(

ubidefeo, 2020-10-01:
@tuxedo0801 the issue has been left open by accident, as it was duplicate of a closed issue.
we periodically groom issues once tasks are completed, but sometimes things are left behind.
we're a small team, it can happen :)

closed 2020-10-01 by ubidefeo