arduino/arduino-pro-ide #160: Can't open pro ide on Ubuntu 19.10

gobbson, 2019-12-03:
Need better instructions to open ide.

Can't seem to open Arduino Pro IDE on Ubuntu 19.10. I'm using the command sudo (location)/arduino-electron to open it, just like many people say to do, but it says that this command doesn't exist.

I'm pretty sure this is because I'm typing something incorrectly, but would be really grateful if someone could answer this.

Thanks for your attention! Have a nice day.

kittaakos, 2019-12-03:
Thanks for reporting it, @gobbson. Do you see any error in the console if you start the application from the terminal? Please copy-paste the full output here. Thank you!

gobbson, 2019-12-03:
hgobbi@hgobbi-Inspiron-3421:~/Downloads/arduino-pro-ide-002/arduino-pro$ sudo arduino-electron
[sudo] password to hgobbi: 
sudo: arduino-electron: command not found

I opened the extracted folder on terminal and typed that code, as told by some forum. Nothing happens. I think I'm trying the wrong command, given. I'm really new to linux so sorry if this is something really stupid.

kittaakos, 2019-12-03:
> I'm really new to linux so sorry if this is something really stupid.

No problem.

So you downloaded the zip, extracted it and if you open a terminal (or navigate to extract location), you can list (`ls -al`) the files.
```
kittaakos@ubuntu:~/Desktop/Arduino Pro IDE-v0.0.2-linux$ ls -al
total 151152
drwxrwxrwx 5 kittaakos kittaakos      4096 Oct 25 15:22 .
drwxr-xr-x 4 kittaakos kittaakos      4096 Dec  3 16:17 ..
-rwxrwxrwx 1 kittaakos kittaakos 114927200 Oct 25 15:21 arduino.pro.ide
-rwxrwxrwx 1 kittaakos kittaakos    167930 Oct 25 15:21 chrome_100_percent.pak
-rwxrwxrwx 1 kittaakos kittaakos    250097 Oct 25 15:21 chrome_200_percent.pak
-rwxrwxrwx 1 kittaakos kittaakos  10221472 Oct 25 15:21 icudtl.dat
-rwxrwxrwx 1 kittaakos kittaakos    255912 Oct 25 15:21 libEGL.so
-rwxrwxrwx 1 kittaakos kittaakos   3152384 Oct 25 15:21 libffmpeg.so
-rwxrwxrwx 1 kittaakos kittaakos   6058008 Oct 25 15:21 libGLESv2.so
-rwxrwxrwx 1 kittaakos kittaakos   7199112 Oct 25 15:21 libVkICD_mock_icd.so
-rwxrwxrwx 1 kittaakos kittaakos      1060 Oct 25 15:21 LICENSE.electron.txt
-rwxrwxrwx 1 kittaakos kittaakos   1993542 Oct 25 15:21 LICENSES.chromium.html
drwxrwxrwx 2 kittaakos kittaakos      4096 Oct 25 15:21 locales
-rwxrwxrwx 1 kittaakos kittaakos    125011 Oct 25 15:21 natives_blob.bin
drwxrwxrwx 3 kittaakos kittaakos      4096 Oct 25 15:22 resources
-rwxrwxrwx 1 kittaakos kittaakos   8691077 Oct 25 15:21 resources.pak
-rwxrwxrwx 1 kittaakos kittaakos    641676 Oct 25 15:21 snapshot_blob.bin
drwxrwxrwx 2 kittaakos kittaakos      4096 Oct 25 15:21 swiftshader
-rwxrwxrwx 1 kittaakos kittaakos   1040440 Oct 25 15:21 v8_context_snapshot.bin
```

The `arduino-pro-ide` is the executable file, so you have to run `./arduino-pro-ide`, like this and it starts:
```
kittaakos@ubuntu:~/Desktop/Arduino Pro IDE-v0.0.2-linux$ ./arduino.pro.ide 
root INFO Theia app listening on http://localhost:45953.
daemon INFO >>> Starting arduino-cli version: 0.6.0 commit: f99361a daemon from /home/kittaakos/Desktop/Arduino Pro IDE-v0.0.2-linux/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Checking for config file in: /home/kittaakos/.arduino15/arduino-cli.yaml
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Candidate IDE Directory: /home/kittaakos/Desktop/Arduino Pro IDE-v0.0.2-linux/resources/app/node_modules/arduino-ide-extension/build
daemon INFO CLI is not bundled into the IDE
daemon INFO Checking for config in: /arduino-cli.yaml
daemon IN
```

If you are wondering why do you need `./` read this thread: https://unix.stackexchange.com/questions/4430/why-do-we-use-to-execute-a-file

Please check if you can run it, if so, please close this issue. If no, let me know what went wrong.

gobbson, 2019-12-03:
Wow, it worked! I needed to run `./arduino-pro-ide`. I was initially trying to open it from the files directly, but I couldn't do it.

Thanks a lot!!
Have a nice day!

closed 2019-12-03 by gobbson

twillert02, 2020-08-16:
_It does not work  on raspbian:_
pi@raspberrypi:~/Arduino Pro IDE-v0.0.6-linux-x64 $ ./arduino-pro-ide
bash: ./arduino-pro-ide: cannot execute binary file: Exec format error

or as root:  somewhere a syntax erroro
pi@raspberrypi:~/Arduino Pro IDE-v0.0.6-linux-x64 $ sudo ./arduino-pro-ide
./arduino-pro-ide: 1: ./arduino-pro-ide: Syntax error: ")" unexpected
pi@raspberrypi:~/Arduino Pro IDE-v0.0.6-linux-x64 $

kittaakos, 2020-08-16:
> It does not work on raspbian:

Thank you for your interest; it's a known issue: https://github.com/arduino/arduino-pro-ide/issues/287

We are working on a solution for this. Stay tuned.

Tex59dk, 2020-09-12:
> _It does not work on raspbian:_
> pi@raspberrypi:~/Arduino Pro IDE-v0.0.6-linux-x64 $ ./arduino-pro-ide
> bash: ./arduino-pro-ide: cannot execute binary file: Exec format error
>
> or as root: somewhere a [...]

The RaspBerry need ARMHF compiled programs the Arduino Pro IDE-v0.0.6-linux-x64 is a X64 compiled program. So, you need to compile the Arduino Pro Ide from source.

twillert02, 2020-09-12:
Thanks now clear. Now to get acces to the source and compliler.

kittaakos, 2020-09-12:
The ARM support currently depends on https://github.com/clangd/clangd/issues/514.

ubidefeo, 2020-09-12:
@twillert02 
we have not released the source code for the Pro IDE yet, so you cannot build it yourself.
Raspian is not 64 bits, but there is a Beta version of Raspberry Pi OS you can download.
Moreover, like @Tex59dk said (thank you),  you would need a version compiled for `aarch64` which is not the Linux64 you can download now.

twillert02, 2020-09-12:
So it is all quite unfortunate, but thanks anyway