arduino/arduino-pro-ide #161: Column Mode (Alt+Drag) for Selection

jamesgohl, 2019-12-08:
A lot of editors these days allow column mode selection by holding ALT down and dragging the mouse over an area of text, this is on VS and Notepad++, it is extremely useful for replacing text.

I would love to see this implemented in the new Arduino IDE :)

kittaakos, 2019-12-09:
It is supported just as VS Code supports it: https://code.visualstudio.com/docs/editor/codebasics#_multiple-selections-multicursor

[image]

Please reopen if required.

closed 2019-12-09 by kittaakos

jamesgohl, 2019-12-09:
> 
> 
> It is supported just as VS Code supports it: https://code.visualstudio.com/docs/editor/codebasics#_multiple-selections-multicursor
> 
> [image]
> 
> Please reopen if required.

What you have demonstrated involves multiple clicks and without the ALT + <drag> functionality I wouldn't have know it existed.

Here is what I mean, the selection rect is possible with ALT key held down ...

[image]

As you can see this is a very powerful editing feature, allowing copy and paste into this explicit selection rect.

And in Visual Studio Professional ...

[image]

kittaakos, 2019-12-09:
I linked the doc, please read. You can use Shift+Alt+Selction: https://code.visualstudio.com/docs/editor/codebasics#_column-box-selection

[image]

jamesgohl, 2019-12-09:
Ahh ok.  Standard is ALT+Drag across all applications.  

VS Code doesn't have this, but VS Professional does - and many others.