arduino/arduino-pro-ide #162: Can't upload sketch to Arduino uno 0.0.3 Ide version

fcebreros, 2019-12-12:
I'm unable to upload a sketch to an Arduino Uno  on arduino-pro-ide 0.0.3

On first click I get an Required Tool message
"
[INFO] Searching tools required for board arduino:avr:uno
[INFO] Required tool [tool: {ToolName:"arduinoOTA",ToolVersion:"1.3.0",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"avr-gcc",ToolVersion:"7.3.0-atmel3.6.1-arduino5",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"avrdude",ToolVersion:"6.3.0-arduino17",ToolPackager:"arduino"}]
[INFO] Adding libraries dir [dir: C:\Program Files (x86)\Arduino\libraries, location: ide]
[INFO] Adding libraries dir [dir: C:\Users\new\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.1\libraries, location: platform]
[INFO] Adding libraries dir [dir: D:\System\Documents\Arduino\libraries, location: sketchbook]
[INFO] Searching tools required for board arduino:avr:uno
[INFO] Required tool [tool: {ToolName:"arduinoOTA",ToolVersion:"1.3.0",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"avr-gcc",ToolVersion:"7.3.0-atmel3.6.1-arduino5",ToolPackager:"arduino"}]
[INFO] Required tool [tool: {ToolName:"avrdude",ToolVersion:"6.3.0-arduino17",ToolPackager:"arduino"}]
[INFO] Tool required for upload [tool: Version: 1.3.0 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
[INFO] Tool required for upload [tool: Version: 7.3.0-atmel3.6.1-arduino5 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin14, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
[INFO] Tool required for upload [tool: Version: 6.3.0-arduino17 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin12, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] Tool required for upload [tool: Version: 1.0.0 Systems: [OS: i686-pc-linux-gnu, OS: x86_64-pc-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: arm-linux-gnueabihf, OS: arm64-linux-gnueabihf]]
[INFO] Tool required for upload [tool: Version: 5.8-arduino11]
"

on second click on Upload I get an Can't open device error.
"
\avrdude: ser_open(): can't open device "\\.\COM4": Access is denied.
Upload error: Error: 2 UNKNOWN: uploading error: exit status 1
"

Windows Version	10.0.18362 Build 18362

MicheleCioccarelli, 2020-03-02:
Have you tried restarting the IDE and your PC?

per1234, 2021-02-18:
>On first click I get an Required Tool message

This was normal. The unnecessary verbosity of the output has since been reduced so you'll only see the useful information with the modern Arduino Pro IDE version.

>on second click on Upload I get an Can't open device error.

This can happen if you click the upload button again while the first upload is still in progress. Unlike the classic Arduino IDE, Arduino Pro IDE doesn't disable the upload button while an upload is in progress.

So I'll go ahead and close this. If you try it out with [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234