arduino/arduino-pro-ide #166: Bug completation

chipxx, 2019-12-18:
Hi, I'm trying, finally, this version that has the autocompletion, arduino-pro-ide-0.0.3-alpha.preview-windows, but I found that it only works once. Let me explain: I run the program I start to write some commands and he immediately suggests the command. I do it the second time and it doesn't work anymore... Does it happen only to me or is it normal?

locotron, 2019-12-30:
Same happens to me on linux and windows, just works for once and then it doesn't

spoenemann, 2020-01-08:
Should be fixed with the next version.

closed 2020-01-08 by spoenemann

Master811129, 2020-01-30:
> Should be fixed with the next version.

I tested 0.0.4 alpha version. the auto-completion is working now but in some cases it is still laggy. Sometimes ,it stops working for a few seconds and then it starts working again.(I think it resets itself. I don't know.) but still good for an alpha version. Keep it up.