arduino/arduino-pro-ide #171: Multiple Serial Connections simultaneously

Hedelinoo, 2020-01-04:
On the old IDE I wished that I could open several Serial Monitors that each display the data of a different Arduino. So this would be great to see in the new one.
Additionally it would be very nice if one could have multiple serial data streams in a single serial plotter. For example sending to variables from 1 or more Arduinos to the serial plotter and have them plotted as to individual graphs.

CreativeRobotics, 2020-11-11:
I want a feature like this as well - At least one extra axilluary serial monitor. The primary serial port would be automatically used for code uploads.
A few other serial features I would love to see:
- Serial file upload with a throttle so the rate at which data is sent from a file can be limited (i.e specify a microsecond delay between bytes)
- Option to log serial data to a file.
- some way of sending characters like TAB and other non typeable ASCII values.

The serial plotter is a very useful feature but it needs improving:
Add an 'auto scale' checkbox so automatic scaling can be turned off.
Add a zoom slider for manually adjusting the display scale.

ubidefeo, 2020-11-15:
@Hedelinoo @CreativeRobotics 
this particular feature is included in the latest nightly build (not in the release yet) which you can download from [here](https://github.com/arduino/arduino-pro-ide#nightly-builds)

Every window (document) has its own serial monitor. Mind that if one serial monitor is connected to a particular board you cannot connect to the same board from another window.

Try it out and let us know what you think.

@CreativeRobotics for the other features requests
> Serial file upload with a throttle so the rate at which data is sent from a file can be limited (i.e specify a microsecond delay between bytes)

what would be the purpose of this?

> Option to log serial data to a file.

This is definitely interesting, we'll have to investigate

> some way of sending characters like TAB and other non typeable ASCII values.

this would require quite a bit of research in terms of UX/UI because right now we only support a text field.
Maybe parsing escaped characters might do the trick.

CreativeRobotics, 2020-11-16:
Thanks for the response. I will try out the multiple serial instances.

For the tab character, how about a 'send value' button with a numeric text entry field - enter an integer and press send to transmit that ascii value ...? You would probably want a UI feedback notification next to it so it displays the actual value sent so if you enter 300 it caps that to 255, and tells you ... Failing that a drop down menu, but that would be a bit clunky with 255 options .

The serial file upload with a throttle comes from a particular use case of mine where I needed to upload a few Kb of data into memory for a look up table (on a SAMD21 based board). The data file was being generated from excel, so it was ascii, not raw data, and the system had to parse each value as it came in whilst also doing a few other things. It turned out to be a bit tricky and I ended up using TeraTerm instead of the serial monitor because it would put a delay between each byte sent. 

Ther are other possible use cases, for example streaming data to a SD card, but I recognise that my need for it may be perculiar.

ubidefeo, 2020-11-17:
@CreativeRobotics 
some of your cases are very edge cases for advanced users who (like me) end up writing tools in Python for :D

CreativeRobotics, 2020-11-17:
> 
> 
> @CreativeRobotics
> some of your cases are very edge cases for advanced users who (like me) end up writing tools in Python for :D

Yes - but not every embedded systems engineer is also a python programmer. The question is, if you are going to provide a serial terminal, then what features should a pro version include?

My corner case may not be a good candidate .... but if you do include a 'send file' option, then maybe its worth considering, and maybe other terminal emulators include it for a reason.

ubidefeo, 2020-11-17:
>Yes - but not every embedded systems engineer is also a python programmer. The question is, if you are going to provide a serial terminal, then what features should a pro version include?

point taken ;)

>but if you do include a 'send file' option, then maybe its worth considering, and maybe other terminal emulators include it for a reason.

I believe there are great serial tools out there that do a much better job (I use `screen` and `Serial`) at all these things, while this (at its current state) is mostly a quick and dirty way of reading output and throwing in strings.
We'll see where future directions take us, and rest assured that we keep track of users' requests 👍

CreativeRobotics, 2020-11-18:
I take your point - in fact I was already realising it when I wrote my last comment ;)

ubidefeo, 2020-11-21:
@CreativeRobotics I just started following a project called [Arduino Cereal Monitor](https://github.com/mladenilic/arduino-cereal#readme) which @rsora sent me a link to.
Looks like fun :)

per1234, 2021-02-18:
As Ubi mentioned, Arduino Pro IDE now has support for an independent board/port/Serial Monitor in each window!

So I'm going to close this as the original feature request being resolved. There are some other tangential requests mentioned here which of course are not resolved. You both are welcome to open dedicated issues to track those requests if you like. Thanks!

closed 2021-02-18 by per1234