arduino/arduino-pro-ide #174: Bug no auto complete

Ferreira31, 2020-01-08:
Quando entro em um novo sketch, o auto complete funciona normalmente, mas em pouco tempo, ele para de completar, e so volta se eu fechar tudo abrir de novo e criar um novo sketch, nao sei se tem algum lugar que tenho q habilitar, se tem nao achei, mas as vezes ta funcionando, outros não.
Agora eu percebi que ele na verdade nao ta reconhecendo os comando, quando digito os comandos, por exemplo:

Serial.begin(9600);
O serial aparece grifado como se nao existisse, e tem em alguns momentos que quando funciona o auto complete, nao aparece todos os codigos certos, por exemplo, digito pin, nao aparece o pinmode

spoenemann, 2020-01-13:
Output of Google Translate:

> When I enter a new sketch, auto complete works normally, but before long it stops completing, and only comes back if I close everything open again and create a new sketch, I don't know if there is any place I have to enable, if I haven't found it, but sometimes it's working, sometimes not.
Now I realize that he is not really recognizing the commands when I type the commands, for example:
> 
> Serial.Begin (9600);
> The serial appears underlined as if it did not exist, and sometimes there is that when auto complete works, not all the right codes appear, for example, type pin, not pinmode

spoenemann, 2020-01-13:
> The serial appears underlined as if it did not exist, and sometimes there is that when auto complete works, not all the right codes appear, for example, type pin, not pinmode

Which board have you selected?

Could you give a code example where you are missing certain proposals in the autocompletion?

Ferreira31, 2020-01-14:
[image]

Ferreira31, 2020-01-14:
[image]

steamhyperpolyglot, 2020-02-04:
Hey guys, not sure if this issue is still open, but after downloading the preview recently and trying it out, I noticed the same thing in my Windows environment.

[image]

per1234, 2021-02-18:
Hi @Ferreira31 and @steamhyperpolyglot . Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-18 by per1234