arduino/arduino-pro-ide #176: Function Tree

Ferreira31, 2020-01-14:
No Outline eu ja vi que aparece a função Loop e a função void, mas agora esta bugada, quando eu inicio, ele está la, quando escrevo qualquer coisa some, não sei até onde vai essa ideia de colocar a arvore de funções, mas ja implementado, seria legal se realmente estivesse todas as funções, voids / if / while e etc, as funções que fecham codigo, e que venha descrição após ela dependendo do comentario que está no codigo.
E que seja possivel minimizar as arvores por funções e etc.
Alem disso que seja possivel minimizar as mesmas em codigo, por exemplo, não quero ver a função setup agora, minimiza ela, e pronto.

[image]

Então por exemplo, na arvore, esse codigo abaixo ficaria da seguinte maneira:
void setup
void test (Description 1)
         if (If test)
void test2 (Description 2)
         if (test2)
void loop

spoenemann, 2020-01-15:
Output of Google Translate:

> In Outline I've seen that appears the Loop function and void function, but now this bug, when I start, it is there, when I write anything goes away, I don't know how far this idea of putting the function tree goes, but already implemented, it would be nice if really were all functions, voids / if / while and etc, the functions that close code, and come description after it depending on the comment that is in the code.
>And that it is possible to minimize the trees by functions and etc.
> Besides that it is possible to minimize them in code, for example, I don't want to see the setup function now, minimize it, and that's it.
>
> So for example, in the tree, this code below would look like this:
> ```
> void setup
> void test (Description 1)
> ------- if (If test)
> void test2 (Description 2)
> ------- if (test2)
> void loop
> ```

Ferreira31, 2020-01-20:
Right, is this