arduino/arduino-pro-ide #177: Problem with search board

Master811129, 2020-01-15:
**Describe the bug**
"Search board"  doesn't search any 3rd party board.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Select board and port'
2. Click on 'SEARCH BOARD'
3. type a non-official board name ''
4. See bug

**Screenshots**
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

Master811129, 2020-01-29:
I think the IDE has some problems with "space" character.
Do you have any idea?

kittaakos, 2020-01-29:
> any 3rd party board.

Can you please share your 3rd party core URL which contains `ESP32 Dev Module`. How did register your 3rd party boards list into the app, did you do it via the CLI manually?

kittaakos, 2020-01-29:
> 3rd party core URL which contains `ESP32 Dev Module`

I guess this one: https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json. At least it contains the `ESP32 Dev Module` board.

Master811129, 2020-01-29:
> > 3rd party core URL which contains `ESP32 Dev Module`
> 
> I guess this one: https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json. At least it contains the `ESP32 Dev Module` board.

I use this 3rd party core
https://dl.espressif.com/dl/package_esp32_index.json

i added this core using JAVA ide.
i'm using 0.0.3 alpha and win10.

kittaakos, 2020-01-30:
It seems to be a CLI issue. I have opened an upstream ticket: https://github.com/arduino/arduino-cli/issues/569

kittaakos, 2020-01-30:
This is a missing feature. See my comment here: https://github.com/arduino/arduino-cli/issues/569#issuecomment-580196238

The Pro IDE is not yet fully support 3rd party boards. Let me check how much effort would it take to implement. Bear with me.

kittaakos, 2020-02-03:
Some progress on this item:

[image]
[image]
[image]
[image]

Please note, it is pretty much in development state and is currently blocked by arduino/arduino-cli#570

wcarvalho0, 2020-02-12:
> Some progress on this item:
> 
> [image]
> [image]
> [image]
> [image]
> 
> Please note, it is pretty much in development state and is currently blocked by [arduino/arduino-cli#570](https://github.com/arduino/arduino-cli/issues/570)

on my arduino pro ide it doesn't have this arduino tab from the second image

kittaakos, 2020-02-12:
> on my arduino pro ide it doesn't have this arduino tab from the second image

We have not released it yet. Work in progress...

Please bear with us.

wcarvalho0, 2020-02-12:
> > on my arduino pro ide it doesn't have this arduino tab from the second image
> 
> We have not released it yet. Work in progress...
> 
> Please bear with us.

ok thank you very much, do you have any prediction of when it will be released?

Master811129, 2020-08-25:
This issue has been fixed so i close it

closed 2020-08-25 by Master811129