arduino/arduino-pro-ide #178: When the debugger will be released?

wanggaoteng, 2020-01-17:
Hi, in this blog:
[https://blog.arduino.cc/2019/10/25/new-arduino-pro-ide-a-closer-look/](url)
it says that:
-------------------------------------------------------------------------------
> Debugger coming soon!
Set breakpoints, view trace, step through execution and much more
Debug your application without affecting timing with a Serial.print()
-------------------------------------------------------------------------------
When this debugger will be released? And can it support arduino uno or mega2560?

Best regards.

spoenemann, 2020-01-20:
Hi! A first preview of the debugger will be included in version 0.0.5 (probably beginning / mid February).

The preview version will not support all boards, but I can't provide a compatibility list yet.

Master811129, 2020-01-20:
> Hi! A first preview of the debugger will be included in version 0.0.5 (probably beginning / mid February).
> 
> The preview version will not support all boards, but I can't provide a compatibility list yet.

and when the 0.0.4 version will be released?

spoenemann, 2020-01-21:
As soon as possible. It's currently blocked by #179.

gmacario, 2020-02-27:
According to the release notes, v0.0.5-alpha (released today) includes debugger support for SAMD boards: https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.5-alpha.preview

closed 2020-02-28 by spoenemann