arduino/arduino-pro-ide #179: Application hangs on startup

spoenemann, 2020-01-20:
**Describe the bug**
In certain situations (on Windows) the application does not start, but shows the spinner animation on startup.

**To Reproduce**
No precise steps to reproduce the problem could be determined, but it seems that unusual characters in the user sketches directory can cause this.

**Expected behavior**
At the least, a meaningful error message should be shown to the user.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10

**Additional context**
It is likely that this `fs-extra` bug is causing the backend to hang while creating a sketch: https://github.com/jprichardson/node-fs-extra/issues/740

spoenemann, 2020-01-23:
Solved.

closed 2020-01-23 by spoenemann

spoenemann, 2020-01-23:
Note: This problem was caused by Windows [Controlled Folders](https://docs.microsoft.com/en-us/windows/security/threat-protection/microsoft-defender-atp/controlled-folders). With the fix included in v0.0.4, the application should no longer hang in the startup phase, but it still won't work as expected because it needs to be whitelisted before it can create anything in `$HOME\Documents\Arduino`.

kbickham, 2020-02-18:
It hangs on start up for me still.

spoenemann, 2020-02-24:
@kbickham please create a new issue and describe with more detail your situation.