arduino/arduino-pro-ide #182: [linux] Unable to start debug session - Error "this.gdbServer.off is not a function"

aentinger, 2020-01-21:
**Describe the bug**
When attempting to start the debugger the error message "this.gdbServer.off is not a function" appears.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to Advanced Mode
2. Create a simple test sketch
3. Compile and download to board to verify programming and downloading works (and view output in serial monitor)
4. Press **Debug -> Start Debugging**.

**Expected behavior**
I would have expected to be launched into some kind of debugging mode where I can single-step the program.

**Screenshots**
The error message is visible in the lower / right corner of the IDE. No output on the console used to launch the Pro IDE.
[image]

**Desktop (please complete the following information):**
 - OS: `Ubuntu 18.04.3 LTS`
 - Pro IDE: `Arduino Pro IDE-0.0.4-3e8c609-linux`

**Additional Information**
- Debugger: `Black Magic Probe V2.0`
- Target Board: `Arduino Zero`

spoenemann, 2020-01-23:
Solved.

closed 2020-01-23 by spoenemann

aentinger, 2020-01-24:
I can confirm that debugging works now for **Arduino Pro IDE-0.0.4-dd0b9bf-linux** :+1: