arduino/arduino-pro-ide #184: Change application settings directory

spoenemann, 2020-01-23:
Arduino Pro IDE stores its settings (`recentworkspace.json`, `settings.json`) in the default location of Theia, which is `$HOME/.theia`. The consequence is that it shares these settings with other Theia-based applications. We should move at least part of these settings to a dedicated directory. For example, it does not make sense to share the recent workspaces with other applications.

akosyakov, 2020-01-23:
It would help if https://github.com/eclipse-theia/theia/pull/6650 is landed.

kittaakos, 2020-03-02:
This is done. Waiting for the review.

For the record, the Pro IDE will store the settings under the `~/.arduinoProIDE/` location.

spoenemann, 2020-03-04:
See https://github.com/eclipse-theia/theia/issues/7218.

kittaakos, 2020-03-04:
> See [eclipse-theia/theia#7218](https://github.com/eclipse-theia/theia/issues/7218).

@spoenemann, see what? What's missing from here?

spoenemann, 2020-03-04:
I just wanted to link that issue as it is related.

closed 2020-05-05 by kittaakos