arduino/arduino-pro-ide #186: Error from gdb after stopping the debugger

spoenemann, 2020-01-23:
**Describe the bug**
The Debug Console shows the following after stopping a debug session:

```
/Users/build/work/GCC-7-build/src/gdb/gdb/thread.c:958: internal-error: int is_thread_state(ptid_t, thread_state): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? 
(y or n) [answered Y; input not from terminal]
This is a bug, please report it.
  For instructions, see:
.
/Users/build/work/GCC-7-build/src/gdb/gdb/thread.c:958: internal-error: int is_thread_state(ptid_t, thread_state): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? 
(y or n) [answered Y; input not from terminal]
Info : dropped 'gdb' connection
```

**To Reproduce**
Start a debug session and then stop it.

**Expected behavior**
Even if the message does not do any harm (the debugger can be restarted normally afterwards), it should not be there. Especially the line _"This is a bug, please report it"_ is quite unfortunate.

boguslawb, 2020-02-28:
What gdb version ? Maybe it's already fixed ? https://sourceware.org/bugzilla/show_bug.cgi?id=12228

spoenemann, 2020-03-04:
Yes, this occurred with an intermediate version.

closed 2020-03-04 by spoenemann