arduino/arduino-pro-ide #192: Toggling Advanced Mode deletes Code

soundstorm, 2020-01-30:
* I just opened 0.0.4 on my Mac, which loaded the first sketch (1) it found in the home directory.
* Opened another file (2) from a subdirectory via Open Icon -> Open... in a new window
* Toggled Advanced Mode in (2) -> On
* Arduino replaces (2) by (1) and now shows two windows with sketch (1)
* Opening (2) via Browse... shows now an empty sketch.

#110 relevant?

spoenemann, 2020-03-05:
I submitted a fix for the "Open..." dialog, available in v0.0.6.

@soundstorm are you sure any code was deleted? That should not happen, and I was not able to reproduce it.

ubidefeo, 2020-03-05:
@spoenemann @soundstorm 
this might be related to #212

kittaakos, 2020-03-30:
The fix is available from the 0.0.6 release.

closed 2020-03-30 by kittaakos