arduino/arduino-pro-ide #193: impossible to close a tab

joysfera, 2020-01-31:
**Bug**
It's impossible to close opened files in tabs. There's no button/widget for that, there's no hotkey (AFAIK) and there's no File->Close menu entry.

**Expected behavior**
A tab has a close widget. Ideally also Ctrl+F4 hotkey works for closing a tab.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 16.04

kittaakos, 2020-01-31:
Thanks for logging, it is intentional. You have to toggle between the _Advanced_ and _Classic_ modes (top right corner on the toolbar). If the _Advanced_ mode is active, then you can close the editors as you wish.

[image]

Please reopen if required.

closed 2020-01-31 by kittaakos

joysfera, 2020-01-31:
Wonderful, thank you! May I suggest you to modify the defaults to let the IDE always start in the _Advanced_ mode then? Just to avoid this super confusion not only I fell into.

kittaakos, 2020-01-31:
> Wonderful, thank you! May I suggest you to modify the defaults to let the IDE always start in the _Advanced_ mode then? Just to avoid this super confusion not only I fell into.

CC: @mastrolinux

dp-cronton, 2020-02-28:
Although you can close a tab in this way, closing a tab is not really an Advanced feature. This should be available in classic mode. (I use Arduino mainly for teaching students aged 17-18 and we would porbably use the classic mode for most students and they should be able to close tabs, when its ready for use.)