arduino/arduino-pro-ide #194: serious SPIFFS bug introduced when compiled by alpha4

joysfera, 2020-01-31:
**Describe the bug**
A working project for ESP32, when compiled by this arduino-pro-ide alpha 4, introduces a serious/scary bug in SPIFFS. It reads content of different file than it opened succesfully.

**To Reproduce**
Steps to reproduce the behavior:
1. Have an existing project with SPIFFS containing some files, say named "A" and "B", containing some data.
2. ```File f = SPIFFS.open("/A", "r"); if (f) { char x[10]; int b = f.readBytes(x, 10); x[b] = 0; Serial.println(x); }```

**Expected behavior**
Shows content of the file "A". Unfortunately, it shows content of file "B" instead.
BTW, it seems to use the real length of file "A" even though it reads bytes of file "B".

Please note this will not be this easy to reproduce. The "A" and "B" are just made up - in fact my files are named "wifi.cfg" and "names.json". It apparently is not related to file names. It might be related on the order of files created on the SPIFFS or their content, not sure.

joysfera, 2020-02-25:
Resolved: what's really happening is that the Arduino Pro IDE is missing the option for selecting the appropriate *Partition scheme*. It seems that I have two different SPIFFS on the flash (by mistake) located at different addresses so if (in Arduino IDE) I select "Minimal SPIFFS" I get one filesystem while if I select "Default 4MB with spiffs" then I get a different filesystem with different files. 

So please add the option for selecting **Partition scheme** to Arduino Pro IDE. Thanks!

Master811129, 2020-02-25:
#10

per1234, 2020-02-29:
Closing as duplicate of https://github.com/arduino/arduino-pro-ide/issues/10. Thanks @Master811129!

closed 2020-02-29 by per1234